paroj/xpad #148: Xbox One (S) controller detected by not working (light is blinking)

Kaldas4, 2020-04-18:
Hello. My Xbox one controller is recognized in jstest and is shaking when i plug it but is definitely not working. When i press the light button, it's blinking. I reinstalled xpad several times but nothing changed...
But what i don't understand is that it worked one time, some hours ago. I think it was after i typed the command i found here : https://www.reddit.com/r/linuxquestions/comments/cvtfbq/getting_my_wired_xbox_one_controllers_to_work_on/ : sudo modprobe xpad

But yet, it's not working again, since i restarted the pc... I'm new to Linux, I don't understand.
I'm using Linux Mint's latest version : 19.3

[image]

I hope someone can help me. Thanks

cgutman, 2020-04-22:
It's probably caused by https://github.com/paroj/xpad/commit/23290cdafd32e573ad814d50fe5683d659bc6918

That init packet doesn't work on older Xbox One S controller firmware.

@paroj It's probably best to revert that commit while we sort out a better solution. It looks like we have a [good candidate](https://patchwork.kernel.org/patch/11500915/). Once we get the finishing touches on it, you can pull it in from Dmitry's input tree.

cgutman, 2020-04-29:
This has been fixed properly upstream.

@paroj Please revert https://github.com/paroj/xpad/commit/23290cdafd32e573ad814d50fe5683d659bc6918 and replace it with https://git.kernel.org/pub/scm/linux/kernel/git/dtor/input.git/commit/?h=for-linus&id=764f7f911bf72450c51eb74cbb262ad9933741d8 when you get the chance.

paroj, 2020-04-29:
done

closed 2020-04-29 by paroj