paroj/xpad #151: SHANWAN Hyperkin Brave Knight gamepad not fully supported in XBOX 360 controller mode

bluesky299, 2020-06-11:
issue context:
when this gamepad is plugged into usb port,  seems it defaults to emulate XBOX 360 controller mode, it briefly identify itself as SHANWAN Hyperkin Game pad,  then it automatically disconnect and reconnect itself as Microsoft X-Box 360 pad

issue:
in evtest,  when pressing any button/key/joystick on gamepad,  no event is registered 
checked online, found info that had to pass dev.ctrl_transfer(0xc1, 0x01, 0x0100, 0x00, 0x14)  to gamepad,  then in evtest,  events are registered, and it works in ps2 emulator pcsx2, vibration work as well

please update xpad to properly support/initialize this gamepad 
looked at xpad source code, seems it make initializations based on vendor:product ids 045e:028e .
with clone controllers impersonate same id 045e:028e , probably need to do additional attribute check on manufacturer, and initialize controllers accordingly

gamepad model info: Hyperkin brave knight premium controller M07304-BK

test os: archlinux v5.4 kernel

dmesg logs
[ 9740.432315] usb 1-1.4: new full-speed USB device number 8 using xhci_hcd
[ 9740.684318] usb 1-1.4: New USB device found, idVendor=2e24, idProduct=16fa, bcdDevice= 2.00
[ 9740.684322] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 9740.684324] usb 1-1.4: Product: Hyperkin Game pad
[ 9740.684326] usb 1-1.4: Manufacturer: SHANWAN
[ 9740.715821] input: SHANWAN Hyperkin Game pad as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1.4/1-1.4:1.0/0003:2E24:16FA.0007/input/input37
[ 9740.716039] hid-generic 0003:2E24:16FA.0007: input,hidraw2: USB HID v1.10 Gamepad [SHANWAN Hyperkin Game pad] on usb-0000:02:00.0-1.4/input0
[ 9740.894169] usb 1-1.4: USB disconnect, device number 8
[ 9741.482294] usb 1-1.4: new full-speed USB device number 9 using xhci_hcd
[ 9741.824470] usb 1-1.4: New USB device found, **idVendor=045e, idProduct=028e,** bcdDevice= 1.10
[ 9741.824474] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 9741.824477] usb 1-1.4: Product: Controller
**[ 9741.824478] usb 1-1.4: Manufacturer: SHANWAN
[ 9741.841647] input: Microsoft X-Box 360 pad as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1.4/1-1.4:1.0/input/input38**
 
-
-
 
below are extra context information about this gamepad , likely have no bearing on above issue
after press HOME button about 8 seconds to switch modes,  mode 2 and mode 3 dmesg logs

[17106.662036] usb 1-1.3: USB disconnect, device number 6
[17106.662296] xpad 1-1.3:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
[17107.199018] usb 1-1.3: new full-speed USB device number 10 using xhci_hcd
[17107.450141] usb 1-1.3: New USB device found, idVendor=2e24, idProduct=16fa, bcdDevice= 2.00
[17107.450145] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[17107.450147] usb 1-1.3: Product: Hyperkin Game pad
[17107.450149] usb 1-1.3: Manufacturer: SHANWAN
[17107.480581] input: **SHANWAN Hyperkin Game pad** as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:2E24:16FA.0008/input/input39
[17107.480769] hid-generic 0003:2E24:16FA.0008: input,hidraw2: USB HID v1.10 Gamepad [SHANWAN Hyperkin Game pad] on usb-0000:02:00.0-1.3/input0

[17153.772393] usb 1-1.3: USB disconnect, device number 10
[17154.308726] usb 1-1.3: new full-speed USB device number 11 using xhci_hcd
[17154.573275] usb 1-1.3: New USB device found, idVendor=2563, idProduct=0526, bcdDevice= 1.00
[17154.573279] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[17154.573282] usb 1-1.3: Product: Android Gamepad
[17154.573284] usb 1-1.3: Manufacturer: SHANWAN
[17154.606346] input: **SHANWAN Android Gamepad** as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:2563:0526.0009/input/input40
[17154.606512] hid-generic 0003:2563:0526.0009: input,hidraw2: USB HID v1.10 Gamepad [SHANWAN Android Gamepad] on usb-0000:02:00.0-1.3/input0
[17154.618341] input: SHANWAN Android Gamepad System Control as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:2563:0526.000A/input/input41
[17154.679293] input: SHANWAN Android Gamepad Consumer Control as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:2563:0526.000A/input/input42
[17154.679388] hid-generic 0003:2563:0526.000A: input,hidraw5: USB HID v1.01 Device [SHANWAN Android Gamepad] on usb-0000:02:00.0-1.3/input1

joaolm, 2021-06-29:
Hi, @bluesky299.

How did you passed the **dev.ctrl_transfer** sequence? And how can I obtain that?

I have a ShanWan controller with the same behavior, it automatically disconnect and reconnect itself as Xbox ONE spectra.
```

[seg jun 28 20:03:49 2021] usb 1-2: new full-speed USB device number 4 using xhci_hcd
[seg jun 28 20:03:49 2021] usb 1-2: New USB device found, idVendor=2563, idProduct=0575, bcdDevice= 1.00
[seg jun 28 20:03:49 2021] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[seg jun 28 20:03:49 2021] usb 1-2: Product: USB Gamepad
[seg jun 28 20:03:49 2021] usb 1-2: Manufacturer: ShanWan    
[seg jun 28 20:03:49 2021] input: ShanWan     USB Gamepad as /devices/pci0000:00/0000:00:01.2/0000:15:00.0/usb1/1-2/1-2:1.0/0003:2563:0575.0003/input/input11
[seg jun 28 20:03:49 2021] hid-generic 0003:2563:0575.0003: input,hidraw2: USB HID v1.11 Gamepad [ShanWan     USB Gamepad] on usb-0000:15:00.0-2/input0
[seg jun 28 20:03:49 2021] usb 1-2: USB disconnect, device number 4

[seg jun 28 20:03:50 2021] usb 1-2: new full-speed USB device number 5 using xhci_hcd
[seg jun 28 20:03:50 2021] usb 1-2: New USB device found, idVendor=24c6, idProduct=542a, bcdDevice= 1.01
[seg jun 28 20:03:50 2021] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[seg jun 28 20:03:50 2021] usb 1-2: Product: Controller
[seg jun 28 20:03:50 2021] usb 1-2: Manufacturer: ShanWan    
[seg jun 28 20:03:50 2021] usb 1-2: SerialNumber: 00007EED872A㐶㔴㘷㐳
[seg jun 28 20:03:50 2021] input: Xbox ONE spectra as /devices/pci0000:00/0000:00:01.2/0000:15:00.0/usb1/1-2/1-2:1.0/input/input12
[seg jun 28 20:03:50 2021] usbcore: registered new interface driver xpad
```

flashrocket, 2021-07-26:
I believe he used https://gist.github.com/dnmodder/de2df973323b7c6acf45f40dc66e8db3 to pass the control sequence. There is also this https://steamcommunity.com/sharedfiles/filedetails/?id=1595753315 which is a step by step procedure to perform the workaround. Would love it if we could somehow do this in xpad itself.

paroj, 2021-08-28:
duplicate of #119

closed 2021-08-28 by paroj