paroj/xpad #157: x360 + RPi Zero = sticks "freezing" apps

roman-bronis, 2020-12-01:
**Description:**
I noticed weird issue when connecting wired Xbox 360 controllers to RPi Zero - all inputs work, but sticks movements "freezes" the controller input events in the apps.

**Steps to reproduce:**
1. Connect controller
2. start app (evtest, jstest) or driver (xboxdrv)
3. watch events coming from controller (they all work)
4. the moment left or right stick is used, for very short amount of time (under 1s) number of events comes from them (x,y nothing suspicious), however then it completely stops and no other event comes after that
5. restarting the application (Ctrl+C and start it again) brings it to working state as seen in point 3, it all works again till point 4. happens again

**More info:**
I tested this with xpad, then with xboxdrv, same behavior. I tested with 2 different Xbox 360 controllers, same issue. I connected the controllers to my desktop PC with Linux and no issue at all. RPi Zero uses latest Raspbian (or whatever they call it nowadays). With xpad I checked dmesg for any clues - nothing. Xboxdrv also shows no errors or warnings.

**Questions:**
Is there known workaround/fix for this?
Is it maybe issue connected to RPi Zero limited resources?
Is it maybe some buffer is full and not cleaning?
Is there a way how to debug this, what information you would need?

Thank you for any help with this. I wanted to use it for cool project for my kid (controlling Lego Duplo train with RPi Zero and old spare Xbox 360 controller).

LAGonauta, 2021-04-20:
I have a similar problem with a RPi 4 and an XBox Series controller connected through USB and an XBox 360 controller connected through a wireless adapter.

However, in my case, those freezes are random and also happen on my Linux desktop. It is really frustrating on racing games...

roman-bronis, 2021-04-20:
@LAGonauta have you tried switchting the driver from xpad to xboxdrv? Give it a shot.