paroj/xpad #161: XBox One S clone controller not initialized

zskamljic, 2021-01-29:
I have a wired ZEROPLUS controller that supposedly supports the original xbox one s protocol, however it looks like it requires an initalization sequence in order to work, as the lights keep blinking after being attached.

lsusb lists it as:
`Bus 003 Device 004: ID 045e:02ea Microsoft Corp. Xbox One S Controller`
which appears to be supported in the source code. If I attach it to a windows machine the blinking stops, as the sequence appears to be sent, however on linux the lights continue to blink. sys/class/leds/xpad*N* does not seem to be created, while /dev/input/js*N* are created. jstest detects the controller, however no keys are detected. 

I have also tried using xboxdrv, however 045e:02ea does not seem officially supported, adding those manually in that source fails as interface class, subclass and protocol dont match, they're 255, 255, 208 in the controller I have, expected values being 88, 66, 0. 

It appears to me that the initialization sequence is not being sent, and as a result the controller is not initialized.

I've attached the capture from USBPcap, the whole communication from attaching the USB device. There has also been a "fix" specified on the arch wiki to resolve this issue: https://wiki.archlinux.org/index.php/Gamepad#Using_generic/clone_controllers_2, however it seems to be specific to that kind of controllers, as changing the vendor and device ids allows me to write the data to the controller and I get a reply from it, but it remains uninitalized.
[output.pcap.txt](https://github.com/paroj/xpad/files/5893789/output.pcap.txt)

zskamljic, 2021-01-30:
I have enabled debug logging, and by the looks of it usb_fill_int_urb fails with enoent:
[ 6037.026458] xpad 3-1:1.0: xpad_irq_in - urb shutting down with status: -2

zskamljic, 2021-01-31:
I've added some extra logs, and additionally enabled DEBUG and DEBUG_VERBOSE, this is what I see when plugging the controller in:

```
[17291.567396] xpad 3-1:1.0: xpad_irq_in - urb shutting down with status: -2
[17291.750757] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17291.750758] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17291.777424] xpad 3-1:1.0: xpad_irq_in - urb shutting down with status: -2
[17291.816687] xpad 3-1:1.0: xpad_irq_in - urb shutting down with status: -2
[17291.979617] xpad 3-1:1.0: xpad_irq_in - urb shutting down with status: -2
[17292.034554] xpad 3-1:1.0: xpad_irq_in - urb shutting down with status: -2
[17292.066756] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17292.066758] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17292.502759] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17292.502762] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17292.938757] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17292.938767] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17293.378759] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17293.378767] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17293.818759] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17293.818767] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17294.258759] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17294.258765] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17294.694761] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17294.694768] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17295.134759] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17295.134767] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17295.574759] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17295.574767] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17296.010761] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17296.010769] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17296.450760] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17296.450762] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17296.890759] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17296.890768] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17297.326761] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17297.326769] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17297.766760] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17297.766768] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17298.206760] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17298.206768] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17298.642761] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17298.642763] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17299.082761] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17299.082763] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17299.518761] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17299.518763] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[17299.958761] xpad-dbg: 00000000: 02 20 01 1c 7e ed 81 64 e0 24 00 00 5e 04 ea 02 03 00 01 00 c5 04 00 00 04 05 01 00 01 00 01 00
[17299.958763] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

zskamljic, 2021-02-05:
I have discovered that my controller requires 2 different packages to be sent, first, as mentioned above:
    
    0x04, 0x20, 0x01, 0x00
Then the following:

    0x01, 0x20, 0x01, 0x09, 0x00, 0x1e, 0x20, 0x10, 0x00, 0x00, 0x00, 0x00, 0x00

Attaching a 2nd controller after that however does not work, only the first one remains active, the 2nd one appears to be stuck in the initializing state.

JoaoGFS, 2021-02-27:
@zskamljic, I have the exact same controller and I'm having the exact same problem. How can I force these packets to be sent to the controller? From what I understand you found a way to make it work, is this correct? Sorry for my ignorance, I just want to get this to work. Thank you!

zskamljic, 2021-03-02:
@JoaoGFS sorry for the late reply, I've attached the patch file I've used. You can follow the "installing" section on the Readme, just make sure to apply it before running dkms install.

[xpad.zip](https://github.com/paroj/xpad/files/6067131/xpad.zip)

davidebeatrici, 2021-04-23:
I have the exact same controller, bought at https://www.ebay.com/itm/254707179803.

@zskamljic Thank you very much for the fix!

Here's the trimmed patch:
```patch
diff --git a/xpad.c b/xpad.c
index cc22bf3..551dbe1 100644
--- a/xpad.c
+++ b/xpad.c
@@ -532,6 +532,15 @@ static const u8 xboxone_rumbleend_init[] = {
        0x00, 0x00, 0x00, 0x00, 0x00
 };
 
+static const u8 xboxone_zeroplus_init1[] = {
+       0x04, 0x20, 0x01, 0x00
+};
+
+static const u8 xboxone_zeroplus_init2[] = {
+       0x01, 0x20, 0x01, 0x09, 0x00, 0x1e, 0x20, 0x10,
+       0x00, 0x00, 0x00, 0x00, 0x00
+};
+
 /*
  * This specifies the selection of init packets that a gamepad
  * will be sent on init *and* the order in which they will be
@@ -543,6 +552,8 @@ static const struct xboxone_init_packet xboxone_init_packets[] = {
        XBOXONE_INIT_PKT(0x0f0d, 0x0067, xboxone_hori_init),
        XBOXONE_INIT_PKT(0x0000, 0x0000, xboxone_fw2015_init),
        XBOXONE_INIT_PKT(0x045e, 0x02ea, xboxone_s_init),
+       XBOXONE_INIT_PKT(0x045e, 0x02ea, xboxone_zeroplus_init1),
+       XBOXONE_INIT_PKT(0x045e, 0x02ea, xboxone_zeroplus_init2),
        XBOXONE_INIT_PKT(0x045e, 0x0b00, xboxone_s_init),
        XBOXONE_INIT_PKT(0x0e6f, 0x0000, xboxone_pdp_init1),
        XBOXONE_INIT_PKT(0x0e6f, 0x0000, xboxone_pdp_init2),
@@ -1835,7 +1846,7 @@ static int xpad_probe(struct usb_interface *intf, const struct usb_device_id *id
                if (usb_endpoint_xfer_int(ep)) {
                        if (usb_endpoint_dir_in(ep))
                                ep_irq_in = ep;
-                       else
+                       else if (usb_endpoint_dir_out(ep))
                                ep_irq_out = ep;
                }
        }
```

sirkhancision, 2021-06-05:
I [forked](https://github.com/sirkhancision/xpad) the repo with the changes made by @zskamljic already pre-made.

joshuamundinger, 2021-06-13:
I tried installing the fork today, and got an error. Here is the make.log:
`DKMS make.log for xpad-0.4 for kernel 5.8.0-55-generic (x86_64)
Sat 12 Jun 2021 09:50:30 PM CDT
make -C /lib/modules/5.8.0-55-generic/build V=1 M=/var/lib/dkms/xpad/0.4/build modules
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-55-generic'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [Makefile:746: include/config/auto.conf] Error 1
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-55-generic'
make: *** [Makefile:5: all] Error 2`
Can someone help me out?

sirkhancision, 2021-06-13:
@joshuamundinger [Found this](https://serverfault.com/questions/568395/what-is-creating-the-generated-autoconf-h) which seems to be related to your issue. Try giving it a go and see if it helps.

lmh, 2021-06-30:
Same controller, same issue. Patch works great. Thanks!
```
sudo git clone https://github.com/sirkhancision/xpad.git /usr/src/xpad-0.4
sudo dkms install -m xpad -v 0.4
sudo rmmod xpad
sudo modprobe xpad
```

paroj, 2021-08-15:
applied a slightly modified version in this branch:
https://github.com/paroj/xpad/tree/zeroplus

please report, whether it also works for you.

jurask, 2021-09-27:
Hi, I had the same issues with zeroplus controller and I can confirm, that patch in zeroplus branch works

whalefood, 2022-01-03:
Really glad I found this thread.  @sirkhancision version worked for my zeroplus controller, but it caused issues with some other controllers I had.  And I didn't have any luck with @paroj's version.  So I made my own fork with some slight alterations that seems to work for all my various knock-off controllers here: [https://github.com/whalefood/xpad](https://github.com/whalefood/xpad)

davidebeatrici, 2022-01-03:
What about my patch? https://github.com/paroj/xpad/issues/161#issuecomment-825369646

whalefood, 2022-01-03:
@davidebeatrici I think your patch is good! The changes I made to sirkhancision's version basically turn it into exactly what your patch is.  I didn't realize the two were different at first

whalefood, 2022-01-03:
Ah.  Only difference is I left this line removed:
XBOXONE_INIT_PKT(0x045e, 0x02ea, xboxone_s_init),

I'll test again with that line back at some point to see if the removal is necessary

jpbrisa, 2022-06-11:
how do i install it on arch linux? sory! first time on linux i use the iso of steamOS code name holo.

Tennyleaz, 2023-05-11:
> applied a slightly modified version in this branch: https://github.com/paroj/xpad/tree/zeroplus
> 
> please report, whether it also works for you.

Yes, this patch is working for me!