paroj/xpad #163: PowerA Blue Camo Xbox One Wired Controller

lemonzest79, 2021-02-09:
Hiya

Just posting a new PowerA Xbox One Controller ID

It is a PowerA Blue Camo Xbox One Wired Controller, ids are 0x24c6, 0x581a

attaching the sudo lsusb -v -d 24c6:581a log

[usb-PowerA-Blue-Camo-Xbox-One-Wired-Controller.txt](https://github.com/paroj/xpad/files/5951873/usb-PowerA-Blue-Camo-Xbox-One-Wired-Controller.txt)

W9JYD, 2021-04-06:
Don't worry about this controller. PowerA support (whom I just complained to) blatantly admitted they will not support Linux or this GitHub repo because...

> Please be advised that our Xbox controllers are compatible with Win 10 PC and Xbox consoles only...

Yeah... Even after telling them about this repo, they say they won't contribute because "proprietary information"... Whatever.

However, [image] a "Retro-Bit Officially Licensed Sega Genesis 8-button Arcade Controller" works. This company (PowerA/BDA) is a Linux/FLOSS-hostile company. Vote with your dollars. Avoid PowerA controllers at all costs.

alex-pek10, 2021-07-18:
Same. I made the addition myself and cannot get it to work. lsusb still showing Thrustmaster.