paroj/xpad #164: SHANWAN PC/PS3 Gamepad - Not working in XInput mode

Unaccounted4, 2021-02-18:
Issue: 
In XInput mode pressing buttons doesn't seem to do anything. No events are shown in fftest or jstest. Switching to DInput reveals the hardware is working and all the buttons show events.

This is present with the default xpad driver and the updated one from this repo.

See details below, let me know if there's anything else you need.

System:    
```
Host: AMiGA-3700X Kernel: 5.10.15-051015-generic x86_64 bits: 64 
           Desktop: KDE Plasma 5.21.0 Distro: KDE neon 20.04 5.21 
Machine:   Type: Desktop Mobo: ASUSTeK model: PRIME X570-P v: Rev X.0x 
           serial: <superuser/root required> UEFI: American Megatrends v: 1407 
           date: 04/02/2020 
Battery:   ID-1: hidpp_battery_0 charge: N/A condition: N/A 
CPU:       8-Core: AMD Ryzen 7 3700X type: MT MCP speed: 3594 MHz min/max: 2200/3600 MHz 
Graphics:  Device-1: NVIDIA GP104 [GeForce GTX 1080] driver: nvidia v: 460.32.03 
           Display: x11 server: X.Org 1.20.9 driver: nvidia resolution: 3840x2160~60Hz 
           OpenGL: renderer: GeForce GTX 1080/PCIe/SSE2 v: 4.6.0 NVIDIA 460.32.03 
Network:   Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet driver: r8169 
Drives:    Local Storage: total: 2.51 TiB used: 1.27 TiB (50.7%) 
Info:      Processes: 371 Uptime: 36m Memory: 15.60 GiB used: 3.39 GiB (21.8%) 
           Shell: bash
```

dmesg: 

```
[ 1725.425386] usb 5-4: new full-speed USB device number 15 using xhci_hcd
[ 1725.583716] usb 5-4: New USB device found, idVendor=2563, idProduct=0575, bcdDevice= 2.00
[ 1725.583717] usb 5-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1725.583718] usb 5-4: Product: PS3/PC Gamepad
[ 1725.583718] usb 5-4: Manufacturer: SHANWAN
[ 1725.636794] input: SHANWAN PS3/PC Gamepad as /devices/pci0000:00/0000:00:08.1/0000:0a:00.3/usb5/5-4/5-4:1.0/0003:2563:0575.000E/input/input44
[ 1725.636859] hid-generic 0003:2563:0575.000E: input,hidraw8: USB HID v1.10 Gamepad [SHANWAN PS3/PC Gamepad] on usb-0000:0a:00.3-4/input0
[ 1725.652636] usb 5-4: USB disconnect, device number 15
[ 1726.309389] usb 5-4: new full-speed USB device number 16 using xhci_hcd
[ 1726.473718] usb 5-4: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.10
[ 1726.473720] usb 5-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1726.473721] usb 5-4: Product: Controller
[ 1726.473721] usb 5-4: Manufacturer: SHANWAN
[ 1726.511730] input: Microsoft X-Box 360 pad as /devices/pci0000:00/0000:00:08.1/0000:0a:00.3/usb5/5-4/5-4:1.0/input/input45
```

Unaccounted4, 2021-02-18:
Seems related to #151 

Running this script after plugging it in initializes it properly: https://gist.github.com/dnmodder/de2df973323b7c6acf45f40dc66e8db3

Is there a way to pass those dev.* bits directly from the driver?

flashrocket, 2021-08-28:
I too own this brand of controller. Would love for the support to be included in driver itself.

paroj, 2021-08-28:
duplicate of #119

closed 2021-08-28 by paroj