paroj/xpad #167: PowerA Xbox One Series X Controller

lemonzest79, 2021-03-21:
I have one of these in Metallic Camo, it's a PowerA Xbox One Series X controller with the ID 20d6:200c the face buttons, sticks and triggers work as does D-Pad, but the Guide (Big X) and the share button show no inputs, I added the line from this patch with my ID changed and it detected the controller, just no inputs from Guide/Share buttons. The controller works in Windows 10 So I'm assuming the controller is fine.

this is the Amazon link for the controller https://www.amazon.co.uk/dp/B08F45381B/

[lsusb.txt](https://github.com/paroj/xpad/files/6050254/lsusb.txt)

I have also noticed the rumble motors are reversed

RenaKunisaki, 2021-09-18:
I have one (wired) that also has two programmable buttons (and a button for programming them), and a mic volume/mute control. Not sure if it's possible for these to be detected as normal buttons. It also doesn't create the /sys/class/leds entry. Otherwise it works fine (except Guide and Share buttons).

fourierules, 2021-10-18:
Im suffering from the same problem. My fusion pro 2 0x20d6 0x4001 has no guide button. The "share" button if I am understanding which button that is.. is in byte 18 bit 0 of the info packet 0x20.

ive tried sniffing an old Microsoft driver which still seems to get the guide button 0x07 messages but i dont see the difference beyond what looks like a long winded conversation over the 0x6 messages. Once those seem to complete there is another power 0x5 message followed by an led message 0x0a before the controller begins sending the battery status 0x03 messages. 
I attempted to duplicate this and didn't find success.