paroj/xpad #170: Non-working: Brook XB Fighting Board - XBOX One mode

HVR88, 2021-08-14:
LSUSB lists device as: **Bus 001 Device 027: ID 045e:0b12 Microsoft Corp. **

JSTEST produces no output when anything is triggered on the joycon. 

Putting the Brook into XBOX360 mode (changes the PID) allows it to operate but the SHARE button comes up the same as the SELECT/VIEW button (#8)

jpoppe, 2022-01-24:
I have the same 'issue' with my Brook Wingman XB adapters which unfortunately can't force x-input mode (mailed Brook support, they say they won't support it for the Wingman XB adapter).

So to use Series S|X mode this driver is required: https://github.com/medusalix/xone

In my case it detects the Wingman XB adapter properly and creates the input device, but it does not register any input from the connected joysticks. I have also asked Brook support about this and they replied they are not interested in adding support for Linux :(.

However the Wingman XB adapter is quite different then the Fighting board (the fighting board does not have to pass any controller) so you could give it a chance ;)

toyoshim, 2022-05-29:
I just found the magic sequence to make it start reporting the input state even in the X-One mode.
The driver needs to send the following packet after the 0x05 start command.
{ 0x06, 0x20, SEQ, 0x02, 0x01, 0x00 }.