paroj/xpad #171: Non working Logitech F710 Wireless gamepad, after the latest update.

ioanx, 2021-08-20:
I have a Logitech F710  wireless gamepad, after the latest update of xpad (commit: 2bfb0e1) the controller doesn't work, it is like i have unplugged it. After i downgrade the driver (commit: 389e258) the controller start to working again.

Machine: Raspberry Pi 4 4gb
Gamepad : Logitech F710  wireless (xinput mode)
Operating System: RetroPie version 4.7.18 (Debian buster Lite 32Bit)
Kernel : Linux retropie 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux

dmesg: 

```
[    9.075512] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    9.075520] usb 1-1.4: Product: Wireless Gamepad F710
[    9.075527] usb 1-1.4: Manufacturer: Logitech
[    9.075534] usb 1-1.4: SerialNumber: 4606946A
[    9.099405] xpad: loading out-of-tree module taints kernel.
[    9.100282] input: Logitech Gamepad F710 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.0/input/input3
[    9.100471] usbcore: registered new interface driver xpad

```

paroj, 2021-08-22:
should be fixed in master now. made the init sequence specific to one controller.

closed 2021-08-22 by paroj

cmitu, 2021-08-23:
Latest commit fails to compile with:
```
...
  gcc -Wp,-MMD,/var/lib/dkms/xpad/0.4/build/.xpad.o.d  -nostdinc -isystem /usr/lib/gcc/arm-linux-gnueabihf/8/include -I./arch/arm/include -I./arch/arm/include/generated  -I./include -I./arch/arm/include/uapi -I./arch/arm/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -mlittle-endian -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -fno-dwarf2-cfi-asm -fno-omit-frame-pointer -mapcs -mno-sched-prolog -fno-ipa-sra -mabi=aapcs-linux -mfpu=vfp -marm -Wa,-mno-warn-deprecated -D__LINUX_ARM_ARCH__=6 -march=armv6k -mtune=arm1136j-s -msoft-float -Uarm -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector-strong -Wimplicit-fallthrough -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -pg -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned  -DMODULE  -DKBUILD_BASENAME='"xpad"' -DKBUILD_MODNAME='"xpad"' -c -o /var/lib/dkms/xpad/0.4/build/xpad.o /var/lib/dkms/xpad/0.4/build/xpad.c
/var/lib/dkms/xpad/0.4/build/xpad.c: In function ‘xpad_start_xbox_360’:
/var/lib/dkms/xpad/0.4/build/xpad.c:1252:52: error: ‘struct input_id’ has no member named ‘device’
  if(xpad->dev->id.vendor != 0x05ac || xpad->dev->id.device != 0x055b)
...
```

Maybe `xpad->dev->id.device` should be `xpad->dev->id.product` ?

paroj, 2021-08-23:
yes! fixed.