paroj/xpad #172: NACON Pro Compact

archanox, 2021-09-14:
I've tried adding to xpad_device `{ 0x3285, 0x0614, "NACON Pro Compact", 0, XTYPE_XBOXONE },` and `XPAD_XBOXONE_VENDOR(0x3285),		/* NACON XBOX Series S/X Controllers */`
But when entering a game it will always spin the camera, as the right thumbstick and the triggers seem to be linked and offsetting each other somehow.

```
T:  Bus=01 Lev=03 Prnt=04 Port=03 Cnt=03 Dev#= 24 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=ff(vend.) Sub=47 Prot=d0 MxPS=64 #Cfgs=  1
P:  Vendor=3285 ProdID=0614 Rev= 1.01
S:  Manufacturer=NACON
S:  Product=Pro Compact
S:  SerialNumber=000001013350E906
C:* #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=47 Prot=d0 Driver=(none)
E:  Ad=01(O) Atr=03(Int.) MxPS=  64 Ivl=4ms
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=4ms
I:* If#= 1 Alt= 0 #EPs= 0 Cls=ff(vend.) Sub=47 Prot=d0 Driver=(none)
I:  If#= 1 Alt= 1 #EPs= 2 Cls=ff(vend.) Sub=47 Prot=d0 Driver=(none)
E:  Ad=08(O) Atr=01(Isoc) MxPS= 228 Ivl=1ms
E:  Ad=87(I) Atr=01(Isoc) MxPS=  64 Ivl=1ms
```

ripper81, 2021-11-27:
I have the same controller. It does nothing with Xpad. I had to install the Xone driver to get it working. While it is detected as an XBox one controller it seems that the device is not implemented into Xpad right now. With Xone the controller ist working. However, Steam only detected it as a generic controller. So i had to set the controller layout to X-Box One Controller and enabled the X-Box Configuration in the general controllersettings in steam. For some games i also had to force or disable the steam controller API in the individual steam game settings to make the controller work as intended.

You should also connect the controller to a Windows PC and install the Nacon App to do firmwareupdate for the device.

slacknk, 2022-05-05:
I'm don't tried this code, but my [Pro Compact Controller](https://www.nacongaming.com/en-GB/pro-compact-controller) don't detected any program. Please add support for this controller

```
[391000.698192] usb 1-1: New USB device found, idVendor=3285, idProduct=0603
[391000.698198] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[391000.698201] usb 1-1: Product: Pro Compact
[391000.698204] usb 1-1: Manufacturer: NACON
```
`dmesg` code from Pro Compact Controller - black (and w/o firmwareupdate). Thanks

ripper81, 2023-05-28:
Are there any news regarding support for this controller?

slacknk, 2024-03-01:
```
{ 0x3285, 0x0614, "Nacon Pro Compact", 0, XTYPE_XBOXONE },
```
Please add this code:
```
{ 0x3285, 0x0614, "Nacon Pro Compact", 0, XTYPE_XBOXONE },
{ 0x3285, 0x0603, "Nacon Pro Compact Black", 0, XTYPE_XBOXONE },
```
I could test it on manjaro

@archanox is your version of the device white-version?

archanox, 2024-03-01:
> @archanox is your version of the device white-version?

Yeah, this one here https://www.jbhifi.com.au/products/rig-nacon-pro-compact-wired-controller-for-xbox-series-xs-and-xbox-one-white

slacknk, 2024-03-01:
Cool, thx. My version [Black](https://www.jbhifi.com.au/cdn/shop/products/520490-Product-0-I-637557379889092784_6501a714-b1ab-46c4-b864-f601d47462a3.jpg), perhaps because of this `ProdID` or `idProduct=` in `dmesg` are different

paroj, 2024-10-08:
fixed with #282

closed 2024-10-08 by paroj