paroj/xpad #174: Turtle Beach Recon Controller not recognised

Keyhaku, 2021-09-27:
Hi,

I've just bought this controller (advertised as xbox one /  xbox series x|s controller) from Turtle Beach : https://www.turtlebeach.com/products/recon-xbox-controller

This is a wired exclusive controller (no wireless).

At first, the device was not even recognised (no led lighten, not powered, not seen in steam, nothing), so I added this udev rule to force loading xpad :
<code>
ATTRS{idVendor}=="10f5", ATTRS{idProduct}=="7005", RUN+="/sbin/modprobe xpad" RUN+="/bin/sh -c 'echo 10f5 7005 > /sys/bus/usb/drivers/xpad/new_id'"
</code>

Now, the device is powered on and is recognised as "generic gamepad", but I can't use it all : 
all the buttons are unresponsive, I cannot assign any button etc.

Can you add support for this device please ?
Is there anything I can do on my side to temporarily make the controller work ?

Thank you for your support

Kernel Info
<pre><code>
[sept.27 10:43] usb 1-9: USB disconnect, device number 8
[  +2,183426] usb 1-9: new full-speed USB device number 9 using xhci_hcd
[  +0,127565] usb 1-9: New USB device found, idVendor=10f5, idProduct=7005, bcdDevice=10.01
[  +0,000013] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000006] usb 1-9: Product: Turtle Beach Recon Controller
[  +0,000005] usb 1-9: Manufacturer: Turtle Beach
[  +0,000003] usb 1-9: SerialNumber: 000001206168D5EF
[  +0,001551] input: Generic X-Box pad as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/input/input33
[  +1,399641] input: Microsoft X-Box 360 pad as /devices/virtual/input/input34
</code></pre>

Lsusb Info
<pre><code>
Bus 001 Device 009: ID 10f5:7005 Turtle Beach Turtle Beach Recon Controller
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71 
  bDeviceProtocol       208 
  bMaxPacketSize0        64
  idVendor           0x10f5 Turtle Beach
  idProduct          0x7005 
  bcdDevice           10.01
  iManufacturer           1 Turtle Beach
  iProduct                2 Turtle Beach Recon Controller
  iSerial                 3 000001206168D5EF
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
</code></pre>

Keyhaku, 2021-09-27:
Ah forgot to mention that I tested this controller on a windows 10 machine, and everything is working fine.
The controller is recognised as a generic xbox 360 controller (HID device xinput compatible).

rarera, 2022-06-06:
Any news on this? How can I make my Turtle Beach Recon Controller work on linux?

closed 2023-01-04 by paroj

sr258, 2023-03-04:
Works with https://github.com/medusalix/xone

DawfukFR, 2023-03-20:
> Works with https://github.com/medusalix/xone

actually my trigger not working with :(

PR #226 (2023-01-04): Added Turtle Beach Recon support
Using the diff from https://www.spinics.net/lists/linux-input/msg75512.html

Of course I don't want to take credit for a diff made by someone else, however it solved my problem that my Turtle Beach Recon controller was not working on Ubuntu 22.04 and other people may find this helpful as well.

It fixes #174