paroj/xpad #175: Gamesir G4 Pro

LukaRusadze, 2021-10-03:
Any change you could add support for Gamesir G4 Pro. On Windows it shows up as Xbox 360 Controller.

lsusb:
`Bus 001 Device 008: ID 8555:061b  Gamesir_G4pro`

barni2000, 2022-01-04:
Hi
I try to add support for it

``` diff
diff --git a/xpad.c b/xpad.c
index cc22bf3..6b73e00 100644
--- a/xpad.c
+++ b/xpad.c
@@ -330,6 +330,8 @@ static const struct xpad_device {
        { 0x24c6, 0x5d04, "Razer Sabertooth", 0, XTYPE_XBOX360 },
        { 0x24c6, 0xfafe, "Rock Candy Gamepad for Xbox 360", 0, XTYPE_XBOX360 },
        { 0x3767, 0x0101, "Fanatec Speedster 3 Forceshock Wheel", 0, XTYPE_XBOX },
+       { 0x8555, 0x061c, "Gamesir G4 Pro", 0, XTYPE_XBOX360 },
+       { 0x8555, 0x061b, "Gamesir G4 Pro Wireless", 0, XTYPE_XBOX360W },
        { 0xffff, 0xffff, "Chinese-made Xbox Controller", 0, XTYPE_XBOX },
        { 0x0000, 0x0000, "Generic X-Box pad", 0, XTYPE_UNKNOWN }
 };

I have installed the driver but it doesn't load, hid_generic loads instead. 

Can you help how can I debug it?

LukaRusadze, 2022-01-04:
Yeah.. I tried that myself but it doesn't seem to do anything for me. 

I'm using xboxdrv right now but it's not perfect considering that you have to do the keymapping yourself and bind it to an event with a dynamic name.  

Also even though the controller has rumble, it doesn't have "ff_rumble" so you can't even get rumble working with xboxdrv.

If you are using the same controller and want to get it working you should look into xboxdrv.

barni2000, 2022-01-04:
xboxdrv is just for remapping for.

For rumble we need xpad.
G4 pro is not emulate xbox 360 gamepad directly the G4s predecessor of G4 pro is using same device and vendor id like the offical xbox360 Gamepad, G4 pro is implementing the xinput protocol but it uses own vendor and device id.

LukaRusadze, 2022-01-07:
I looked into it more. 

From what I gather after adding the vendor id to `xpad_table[]` the driver should try to initialize the controller when plugged in but from what I see even when adding` XPAD_XBOX360_VENDOR(0x8555)`, to the list it still gets loaded as hid-generic gamepad and doesn't even get to the probing part.

From my understanding that means that the driver for some reason fails to even detect the controller, so it doesn't even get to the probing part.

I was able to get the controller to use xpad driver in one situation, when I added `{ USB_DEVICE(0x8555, 0x061b) }` to the `xpad_table[]` and even then I had to disable `usbhid` module. Even though `lsusb -t` reported driver as xpad it didn't seem to have a /dev/input/jsX file so it didn't work anywhere. A few seconds after disabling `usbhid` my pc would freeze up so I couldn't really get any further

Would love your help @paroj since you understand this driver way better than I do.

pablomancera, 2022-01-11:
I bought the same gamepad and looked a bit into it some time ago, I got stuck because I don't know how, the gamepad identifies itself as a Xbox 360 Controller in Windows, but as a generic HID controller in Linux.

[image]

barni2000, 2022-01-11:
@pablomancera Can you send the full pcap from windows? maybe we should check the initialization process.

LukaRusadze, 2022-01-11:
I may not be pablomancera but I did capture me plugging in the controller. Github doesn't allow uploading the pcap file directly so I uploaded it on my google drive.

https://drive.google.com/file/d/1DannmLpdyYtp0nVyr7uJQmeUf0tmAKyh/view?usp=sharing