paroj/xpad #192: Feature Request - Allow triigers and dpad to be seen as buttons and not axis.

adamfmatt, 2022-02-27:
Xpad is great thank you for your work.

Currently Xpad detects triggers/dpad as axis. Of course this is welcome in many games. However there are some situations where having just 4 axis and more buttons would be desired when using in-game mapping tools. It would be quite convenient to have the ability to select this configuration through Xpad.

Thanks again.

adamfmatt, 2022-02-27:
Upon researching this topic there appears to be xpad module parameters that could enable this behavior, yet are restricted to "Unknown Devices". If this was allowed on all devices and enabled with `options xpad dpad_to_buttons=1 triggers_to_button=1` this could be a solution. 

I may try to recompile `xpad` as described here
https://sites.google.com/site/sbobovyc/home/guides/xbox-one-controller-for-x3-albion-prelude

paroj, 2022-03-11:
general button remapping is out of scope for a kernel driver. the kernel option is merely provided to get unknown devices running.

closed 2022-04-19 by paroj