paroj/xpad #194: Logitech F310, doesn't work correct after the last update.

gdniw, 2022-03-10:
I have a Logitech F310 gamepad, after the latest update of xpad (commit: 154c208) some controllers buttons doesn't work in game(A,B,X,Y, START,SELECT). The only buttons that are working are the D-pad and L,R. After i downgrade the driver (commit: bc930a5) the controller start to working again.

Machine: Raspberry Pi3 3B+
Gamepad : Logitech F310 (xinput mode)
Operating System: RetroPie version 4.7.21 (Debian Buster Lite 32Bit)

Fmstrat, 2022-03-10:
@gdniw the latest commit added support for `BTN_1` - `BTN_4` on Elite controllers, and my guess is the F310 is may be sending the same codes for ABXY as those buttons. It may be that we need to separate out the Elite controller bindings from other XBox One style bindings. Are you familiar with turning on `DEBUG` mode and providing some output?

gdniw, 2022-03-10:
> @gdniw the latest commit added support for `BTN_1` - `BTN_4` on Elite controllers, and my guess is the F310 is may be sending the same codes for ABXY as those buttons. It may be that we need to [...]

I am not very familiar. I try to follow the instructions on the first page, i run first `dmesg --level=debug --follow and then i open  a new terninal and i run succesfully the jstest /dev/input/js0 but nothing appear on the log.
`

gdniw, 2022-03-11:
It looks that the change affect more controllers than F310. I have the same issue with _Logitech F710_ and there two reports on the RetroPie forum with the same problems with _8bitdo SN30 Pro Wired_ and _Xbox 360 controller_'
Thanks

paroj, 2022-03-11:
I moved the faulty commit to a separate branch (xpadone_paddles) and reset master until this is resolved.

seems `xpad_common_btn` is the wrong place to add controller specific buttons.

Fmstrat, 2022-03-11:
Smart move, wish I had more devices to test with. I will address this by adding a separate custom button section.

kakra, 2022-03-12:
> @gdniw the latest commit added support for `BTN_1` - `BTN_4` on Elite controllers, and my guess is the F310 is may be sending the same codes for ABXY as those buttons. It may be that we need to [...]

I'm the developer of xpadneo.

First and most important: the default Elite profile aliases those paddles to A,B,X,Y - you'd need to switch to profile 1, 2, or 3, to actually get those reported separately by the controller - at least in HID mode but I suspect it's the same in GIP mode (USB, donlge).

But nonetheless important: The driver is not allowed to use `BTN_1`..`BTN_4`. These have lower keycodes and will be ordered before the expected button map, and thus invalidate all assumptions user space makes. Also, they are from the `BTN_MISC` range and thus identify a misc input device. User-space uses a heuristic where it checks for the first button to device whether a device is a joystick (`BTN_JOYSTICK`), a gamepad (`BTN_GAMEPAD`), a mouse (`BTN_MOUSE`) or some other device (`BTN_MISC`). You are not allowed to mix button maps from different categories, otherwise the heuristic breaks. There are few exception when it is allowed for some device to overflow keycodes into the next category. Instead, for this purpose, the `BTN_TRIGGER_HAPPY` category should be used. As far as I understand, it is allowed to be used as an overflow category for joysticks and gamepads with many buttons. And they will order behind the well-known gamepad mappings which is important.

Here's some thought about it and historical background: https://github.com/atar-axis/xpadneo/issues/286

paroj, 2022-03-14:
@kakra I read if we use `BTN_TRIGGER_HAPPY5..BTN_TRIGGER_HAPPY8` instead of `BTN_1..BTN_4` we should be fine?

kakra, 2022-03-14:
Yeah, that's what I suspect. It should be tested, tho. Also, some additional thought should be put into that regarding future updates to the mapping with new controllers. We already have a Share button which doesn't really match any game input button in Linux. So it should probably put into the same category.

At this point it may make sense to expose all buttons in the input mapping even if that particular model does not support it. This makes live easier for user-space layers like SDL because they don't need to add an additional mapping for each model. IOW, the positional meaning of a button should be kept in place: BTN_A should always be at position 1, B at 2 etc. I'm currently considering a similar model for xpadneo.

paroj, 2022-03-14:
> This makes live easier for user-space layers like SDL because they don't need to add an additional mapping for each model.

I dont think this is particularly important. Their API provides `SDL_CONTROLLER_BUTTON_B`, which they need to remap anyway given this:
[image]

also, the xpad driver exposes the DPAD on the x360 differently, depending whether it wireless or not. Yes, this is insane - but cleaning it up will mean breaking SDL..

kakra, 2022-03-14:
Yep, touching existing behavior is problematic. SDL should really look at evdev symbols instead of button positions. This could eliminate the need for a lot of mapping entries. The Xbox controllers usually provide a proper evdev symbol mapping in HID mode even without special drivers. Non-HID drivers like xpad should try to achieve something similar, maybe even try matching the symbolic mapping of the HID profile. For existing buttons, it's too late. But future updates should try adding new buttons only behind the well-known positions so it doesn't conflict with existing user-space.

Position is dictated by the event code number of the kernel: They are simply ordered by code number by walking the bits in the button map. This is exactly how jsdev works. And SDL uses the same scheme. The SDL controllerdb is just a conversion table from bit position to symbol name. Thus they are reverting what the kernel did (the HID layer converts HID bitmap positions to symbols, SDL "converts" that back to positions, and then to SDL event symbols).

Fmstrat, 2022-03-14:
> @kakra I read if we use `BTN_TRIGGER_HAPPY5..BTN_TRIGGER_HAPPY8` instead of `BTN_1..BTN_4` we should be fine?

I will give this a shot in the new update. I've already limited it down to just within the Elite controllers identifier (just haven't pushed/fully tested yet). Thanks everyone.

medusalix, 2022-03-14:
Keep in mind that the paddle bits can be in different positions based on the firmware version of your controller:

https://github.com/libsdl-org/SDL/blob/7c421fec16ccdc7cfb44b7f9120469d5e24db6e5/src/joystick/hidapi/SDL_hidapi_xboxone.c#L534-L545

kakra, 2022-03-14:
AFAIR, at least in HID mode, the paddle bits will activate the same time when button A,B,X,Y is pressed - while in profile 0. So in profile 0, the paddle bits should be masked so a game doesn't see two buttons virtually depressed at the same time.

Fmstrat, 2022-03-15:
> Keep in mind that the paddle bits can be in different positions based on the firmware version of your controller:
> 
> https://github.com/libsdl-org/SDL/blob/7c421fec16ccdc7cfb44b7f9120469d5e24db6e5/src/joystick/hidapi/SDL_hidapi_xboxone.c#L534-L545

> AFAIR, at least in HID mode, the paddle bits will activate the same time as if button A,B,X,Y is pressed - while in profile 0. So in profile 0, the paddle bits should be masked so a game doesn't see two buttons virtually depressed at the same time.

Agreed, the paddle activation in the original PR is for profile 0, when all three lights are off on the controller. This is how most users use the Elite on PC for mapping and/or remapper/reWASD. I'll have a new PR tomorrow most likely.

paroj, 2022-03-29:
@gdniw can you test again with current master? we updated the paddles code so it should no longer interfere with other gamepads

gdniw, 2022-03-31:
> @gdniw can you test again with current master? we updated the paddles code so it should no longer interfere with other gamepads

I test it and it's working fine. Many many thanks !

closed 2022-03-31 by gdniw