paroj/xpad #198: Nacon GC-100 Recognized but most of the games do not see the controller.

Jeronimo17, 2022-04-18:
Hello,
I just received this controller, which looks very good for the price, I have seen that jstest-gtk recognizes it and it works, Parsec also recognizes it, Steam also but other GOG and non-GOG games that work with an Xbox imitation (0x045e, 0x028e) do not recognize any connected controller.

why?

In lsusb 2 appear:
```
Bus 001 Device 005: ID ffc0:0040 Nacon GC100
Bus 001 Device 008: ID 3285:0607 Nacon GC100
```
I have modified the xpad.c to imitate the one that works and is recognized as original, but it doesn't work either, I have also changed the names of the symbolic links in /dev/input/by-id to copy the name of the "original" and it is not recognized by the games either.
MOD:
[image]
Original:
[image]

`/dev/input/by-id $ ls`
```
usb-Nacon_GC100_JC--8267A_A8-event-joystick
usb-Nacon_GC100_JC--8267A_A8-joystick
```

`/dev/input/by-path $ ls`
```
pci-0000:00:14.0-usb-0:2:1.0-event-joystick             pci-0000:00:15.0-platform-i2c_designware.0-mouse
pci-0000:00:14.0-usb-0:2:1.0-joystick                   platform-clevo_xsm_wmi-event
pci-0000:00:14.0-usb-0:3-event                          platform-i8042-serio-0-event-kbd
pci-0000:00:14.0-usb-0:6.2:1.0-event-mouse              platform-i8042-serio-2-event-mouse
pci-0000:00:14.0-usb-0:6.2:1.0-mouse                    platform-i8042-serio-2-mouse
pci-0000:00:15.0-platform-i2c_designware.0-event        platform-INT33D5:00-event
pci-0000:00:15.0-platform-i2c_designware.0-event-mouse
```

Thank you

paroj, 2022-04-19:
game controller mapping is handled by https://github.com/gabomdq/SDL_GameControllerDB or the game itself

closed 2022-04-19 by paroj

Jeronimo17, 2022-04-19:
So you mean other controllers from other brands copy the xbox button mapping and this one brings its own and the games think there is no controller?
And there is no way to make the games see a normal controller?

This is the first time this has happened to me with a game controller on linux, even the Logitech F310 has worked fine for me. :confused:  

Thanks

paroj, 2022-04-20:
if you take a look at https://github.com/gabomdq/SDL_GameControllerDB/blob/master/gamecontrollerdb.txt, you will see that the F310 is listed for Linux, while the GC100 is not. So games fall back to some arbitrary keymapping.