paroj/xpad #225: 8BitDo Ultimate contoller initialization error

scapeg0at, 2022-11-22:
I've installed the newest patch of xpad to use my 8BitDo Ultimate wireless controller. I'm attempting to connect via the 2.4g adapter but I'm running into what seems like an initialization error. 

When I turn on the controller it vibrates like it is connecting, I can't get any input to register, then the controller shuts off. This all happens in a couple of seconds.

The output of dmesg looks like this
`[ 1127.151124] usb 2-1.1: USB disconnect, device number 16
[ 1127.891837] usb 2-1.1: new full-speed USB device number 17 using ehci-pci
[ 1128.014805] usb 2-1.1: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.14
[ 1128.014814] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1128.014819] usb 2-1.1: Product: Ultimate Wireless Controller
[ 1128.014822] usb 2-1.1: Manufacturer: 8BitDo
[ 1128.014825] usb 2-1.1: SerialNumber: 517cc0d817e4
[ 1128.015954] input: 8BitDo Ultimate Wireless Controller as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/input/input17
[ 1132.271234] usb 2-1.1: USB disconnect, device number 17
[ 1132.271392] xpad 2-1.1:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
[ 1133.007658] usb 2-1.1: new full-speed USB device number 18 using ehci-pci
[ 1133.130984] usb 2-1.1: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.14
[ 1133.130994] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1133.130998] usb 2-1.1: Product: Ultimate Wireless Controller
[ 1133.131001] usb 2-1.1: Manufacturer: 8BitDo
[ 1133.131004] usb 2-1.1: SerialNumber: 517cc0d817e4
[ 1133.132719] input: 8BitDo Ultimate Wireless Controller as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/input/input18
[ 1133.295345] usb 2-1.1: USB disconnect, device number 18
[ 1133.295514] xpad 2-1.1:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
[ 1134.035586] usb 2-1.1: new full-speed USB device number 19 using ehci-pci
[ 1134.149213] usb 2-1.1: New USB device found, idVendor=2dc8, idProduct=3109, bcdDevice= 2.00
[ 1134.149223] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1134.149227] usb 2-1.1: Product: Ultimate Wireless Controller
[ 1134.149231] usb 2-1.1: Manufacturer: 8BitDo
[ 1134.149234] usb 2-1.1: SerialNumber: 517cc0d817e4
[ 1134.158255] hid-generic 0003:2DC8:3109.0006: hiddev0,hidraw0: USB HID v1.11 Device [8BitDo Ultimate Wireless Controller] on usb-0000:00:1d.0-1.1/input0
`
I think the most important line of this output is 
`[ 1132.271392] xpad 2-1.1:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19` 
but I'm not sure what's happening here

The controller also appears in /dev/input/by-id for the brief moment it is connected but it doesn't register any input.

Any help would be appreciated, thank you.

scapeg0at, 2022-11-23:
I realized that the issue wasn't necessarily with the xpad driver, there wasn't a program to hook in with the controller so it simply continued to disconnect.

trecuu, 2023-03-21:
Sorry to revive... Im having the exactly same problem that you have when trying to run my controller on raspberry pi running retropie. How did you solve it?

scapeg0at, 2023-03-22:
Sorry for closing the issue without giving a full explanation of my solution.

What I ended up finding out was that, even if though the xpad driver was installed correctly if I didn't have steam open to "capture" the input from the controller the connection would continuously loop as described above.

What I found interesting was that I didn't have to have the focus on steam, all I needed was to have steam open when I connected via the 2.4 adapter. After the connection I could tab out of steam and still use the controller for input. I believe that there are settings within steam that lets you map a controller for use outside of steam.

It's not an actual solution to the connection cycling but it worked for my use case so I decided to close the issue.

trecuu, 2023-03-23:
Thx for the reply!

I managed to fix my problem, that was exactly the same. But I was using the Bluetooth Ultimate, and OS was retropie.
I had to download a beta firmware, and change to D-Input to be able to work. Even though xpad is expected to work with X-Input.

Anyway... you can close it! 

Thx again for the help!

closed 2023-03-23 by scapeg0at