paroj/xpad #227: 8BitDo Ultimate Wired Controller fails to connect but uses a recognised product ID

VanishedOne, 2022-12-16:
This is about the 8BitDo Ultimate Wired Controller at https://www.8bitdo.com/ultimate-wired-controller/ -- not the 8BitDo Ultimate Wired Controller for Xbox from #217.

It's supposed to support an XInput mode; the manual doesn't explain how to activate it, but what I read online suggests holding down X while connecting the controller, and if I do that it tries to connect with a product ID of 3106. 

Now it seems that the same combination of vendor and product IDs already exists in xpad's list of devices, but for a wireLESS controller:

`{ 0x2dc8, 0x3106, "8BitDo Ultimate Wireless Controller", 0, XTYPE_XBOX360 },`

8BitDo presumably reused the product ID. However, the wired controller doesn't get detected as the wireless one; it fails to connect altogether in XInput mode.

VanishedOne, 2022-12-16:
After a look through some other issues, I think mine may actually be the same as #206.

paroj, 2024-10-08:
fixed with #230

closed 2024-10-08 by paroj