paroj/xpad #228: New Ultimate code conflicts with 8Bitdo Wireless USB 2

btegs, 2022-12-18:
Output from lsusb on Pop!_OS 22.04 with the 6.0.6 kernel and the recent paraoj/xpad:

NOTE: This is the Ultimate 2.4g wireless Controller and the latest firmware for it along with the receiver was installed today: https://www.8bitdo.com/ultimate-2.4g-wireless-controller/

**USB Wireless Adapter 2:**
No connected controller: `Bus 001 Device 011: ID 2dc8:3107 8BitDo IDLE`
Connected an 8Bitdo Pro 2: `Bus 001 Device 012: ID 2dc8:3106 8BitDo 8BitDo Receiver`
Connected an Xbox Elite v2: `Bus 001 Device 018: ID 2dc8:3106 8BitDo 8BitDo Receiver`

**Ultimate Wireless 2.4g Receiver:**
No connected controller: `Bus 001 Device 013: ID 2dc8:3109 8BitDo IDLE`
Connected controller: `Bus 001 Device 014: ID 2dc8:3106 8BitDo 8BitDo Ultimate wireless Controller for PC`

It seems like their is a conflict with the new 8Bitdo Ultimate wireless settings in xpad.c. The Pro 2 controller didn't have a conflict as it is considered an Xbox One controller but the Ultimate only works as an Xbox 360 controller.

If you remove all of the Ultimate related code, the 8Bitdo Wireless USB Adapter shows up as an Xbox 360 controller under these settings:

```
{ 0x045e, 0x028e, "Microsoft X-Box 360 pad", 0, XTYPE_XBOX360 },
```

jcotton42, 2022-12-19:
I'm not an expert on this, so the maintainers will have to weigh in, but it looks like xpad should handle that by matching on protocol and subclass numbers that vary between Xbox 360 and Xbox One controllers https://github.com/paroj/xpad/blob/master/xpad.c#L456

I unfortunately don't have the controllers you mention to test, the only 8bitdo I own is the Ultimate Bluetooth Controller, and the dongle that ships with it.

btegs, 2022-12-19:
@jcotton42 - I have the USB Wireless dongle (v1 and v2), the Pro 2, and the Ultimate 2.4ghz controller + dongle. If you want me to run some commands in the terminal or modify the source files and build, let me know.