paroj/xpad #232: GameSir G7 support

JohnTheCoolingFan, 2023-01-13:
Hi, I  have a GameSir G7, probably an xbox one controller ("Wired controller for xbox" is written on the box), it works on windows but doesn't work on linux with xpad.

Here's info about it from lsusb: `ID 3537:1001 Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-G7 Controller for Xbox`

barni2000, 2023-04-04:
Hi @JohnTheCoolingFan
I think this gamepad compatible with https://github.com/medusalix/xone
They add the vendorId at january https://github.com/medusalix/xone/commit/24674076034e21d99e5e09f9b5fbbeaf21a5f8b4

JohnTheCoolingFan, 2023-04-04:
Yes, iirc they added upon my request on discord and I've been using this driver since

loloxploit, 2024-05-31:
thanks guys working on pop o.s here appreaite it!

ArtisticOwl, 2025-02-16:
do you guys where I can get support for my wired g7 se for mac?

IAmJoker47, 2025-03-12:
Im having this problem with my G7 SE. On Arch based distros or just Arch, the controller keeps disconnecting and reconnecting and the xone drivers don't seem to work as it need the "linux-headers". Any solution to this issue?

JohnTheCoolingFan, 2025-03-12:
@IAmJoker47 what do you mean by "don't seem to work as it need the 'linux-headers'"? Is there some error installing the package or from the driver?

IAmJoker47, 2025-03-12:
First of all on Arch, the controller keeps disconnecting and reconnecting.

[image]

I've just been able to solve this by installing xone from the AUR but it's still not working. The LED doesn't light up and I cant do anything with the controller. Tried restarting my pc a couple of times but still not working.

JohnTheCoolingFan, 2025-03-12:
Can't help you much more, G7 SE is a different device. I recommend going to the xpadneo discord server, they have a channel about xone. Tell them about your issue, they might help you with it. There's also some messages before already asking about G7 SE, you can use channel search to find those, they have been answered. In each case there are different things involved.

IAmJoker47, 2025-03-12:
I'm sorry for taking your time. I've finally been able to resolve my problem. Turns out I only needed to install the xone-dkms-git package. Now my controller works perfectly fine.

IAmJoker47, 2025-03-12:
I kinda fixed it while i was in the middle of replying to you soooooo yeah. Couldn't show you the message even I wanted to. Anyway thank you for your time.