paroj/xpad #234: automate kernel patches

colemickens, 2023-01-24:
>actually this is synced about once a year with mainline. Its just a PITA to push patches through the mailing list workflow used by the kernel.

I don't want to come across demanding or ungrateful but this is a pretty high-friction burden on the ecosystem around linux and xpad.

As a new owner of an 8bitdo Ultimate Controller, I'm looking at 6.2 tree and a (not-present out-of-tree) commit to xpad from Nov 2022 and going "wow, this is too bad".

While I theoretically and technically could, I'm not going to package a dkms version of xpad for nixos.

What can "we" do about this? This is a naive question, so please forgive it the imperfections than come with that, but what needs to change? Are there major distros that ship xpad as  dkms? by default?

colemickens, 2023-01-24:
(for example, do I have to disable in-built xpad and then add this module, or?)

paroj, 2023-01-24:
if you follow the steps from the readme, DKMS will take care of integrating with your kernel and replace the built-in module. No need to package anything.

colemickens, 2023-01-24:
this means that the package maintainer has to ensure this package stays up to date, lest it regress changes that might land upstream. Also, it seems out of tree kernel modules have esoteric build processes, so I'm having to figure out how to build xpad now (like, the Makefile using uname to determine what kernel to build for does not bode well for building xpad easily for nixos). Given nix, it would honestly be easier to diff this repo against upstream, and patch the kernel and rebuild it.

paroj, 2023-01-26:
I have now actually read what nixos is about, so I better understand your problems with DKMS.

Maybe this will help in your case as well? https://github.com/paroj/xpad/pull/185/commits/de37b6f97af60229b8e0f421ee1c6414477c5f77

Anyway.. this is a wontfix from my side as DKMS works best for me and I want to stick to a single install method

closed 2023-01-26 by paroj

colemickens, 2023-02-04:
In NixOS, we simply don't use or have any use for DKMS whatsoever. I have this building with our modules build system, using a similar technique as the Gentoo patch you linked earlier.

However, again, since we don't use DKMS, we compose the initramfs with the builtin modular kernel modules, and then any "extra" modules, like in this case, the updated `xpad`.

These, of course, conflict. Do you have any recommendations? I'm inclined to "rename" this to "xpad-tip" or "xpad-dkms" so that it can be addressed separately, the user could presumably blacklist `xpad` and then include `xpad-tip` or whatnot.

But maybe you have another suggestion. Thanks!

colemickens, 2023-02-07:
For now I'm maintaining a patched version of xpad that has a better Makefile, and is named differently, thus allowing me to blacklist in-built and supply `xpad-tip`. I will likely bring this up as a new nixos addition, barring other suggestions.