paroj/xpad #236: Trigger rumble support on Xbox One controllers

motolav, 2023-03-26:
There was someone that submitted patches to the kernel but they ended up unnoticed.
It's not like xpadneo's implementation of trigger rumble but allows userspace to control the trigger rumble motors

These patches require changes outside of xpad.c to `include/uapi/linux/input.h` and `drivers/input/ff-memless.c`

[[PATCH 0/4] input: Implement an API for trigger rumble motors](https://lore.kernel.org/lkml/20220410220449.5071-1-dbdaniel42@gmail.com/)
[[PATCH 1/4] input: uapi: Add trigger_left and trigger_right to ff_rumble_effect struct](https://lore.kernel.org/lkml/20220410220633.5235-1-dbdaniel42@gmail.com/)
[[PATCH 2/4] input: ff-memless: Add trigger left/right in ml_combine_effects](https://lore.kernel.org/lkml/20220410220649.5258-1-dbdaniel42@gmail.com/)
[[PATCH 3/4] input: xpad: Implement trigger rumble for Xbox One controllers](https://lore.kernel.org/lkml/20220410220701.5279-1-dbdaniel42@gmail.com/)
[[PATCH 4/4] input: hid-microsoft: Implement trigger rumble for Xbox One S over bluetooth](https://lore.kernel.org/lkml/20220410220713.5303-1-dbdaniel42@gmail.com/)

paroj, 2023-03-26:
invalid as

> These patches require changes outside of xpad.c

motolav, 2026-05-02:
Closing this as there is a new driver for GIP supported controllers in development that will remove Xbox One controller support from xpad.
[[PATCH v3 00/10] Input: xbox_gip - Add new driver for Xbox GIP](https://lore.kernel.org/all/20260310052017.1289494-1-vi@endrift.com/)