paroj/xpad #242: feature request - Xbox 360 Chatpad support?

tomeq82, 2023-05-06:
Hi,

Is there any interest or possibility of making Xbox 360 Chatpad usable under Linux? Or i'm the only one owner who find it quite usable for HTPC and other gaming setups? :) 

Is this hardware anyhow artificial? I have this chatpad and I can easily make some debugs/dumps but I need to know how and what can be of interest....

UnnamedUser3000, 2023-06-28:
I have two of them. It would be so nice if i could use them instead of having them gather dust on a shelf. I have been trying to get an alternative driver, xboxdrv, working with no luck. they only work for wired controllers, witch I don't have.

paroj, 2024-10-14:
pull-requests are welcome

segakawaii, 2024-12-17:
Would really love this feature, as I prefer using Wireless Xbox 360 controllers on my HTPC. The old xboxdrv driver states that it supports the chatpad, but I was not able to get it to work. However, I was able to get the chatpad to work using the fork of xboxdrv by zerojay, but only with the wired controller. I was not able to figure out a way to get it to work with the wireless controllers. Not really sure if anything in that project could be useful for figuring out how to use it in this one, but I wanted to mention it here just in case.

blager, 2026-11-13:
For anybody interested in tackling this, it might help to know that it seems that the communication part is solved.
- `xboxdrv` has recently added an experimental support for wireless variant, together with updating docs
- the linked #367 is usable