paroj/xpad #244: xbox series elite 2 select button doesn't work on bluetooth - but does in xpadneo

fbodymechanic, 2023-05-18:
I've tried this and have it working on Fedora 37 and now 38, and I used the AUR package xpadneo when on Arch/based distros.  When I have the xbox series elite controller connected via bluetooth on the xpad drivers when I connect, the control doesn't do the vibration and the select button - 2 squares middle center on controller do not work without the xpadneo package.  I was told I should report this to you.  I don't know what else you would need from me.  Thanks!

I have tried and verified it works with these packages:

https://aur.archlinux.org/packages/xpadneo-dkms

https://copr.fedorainfracloud.org/coprs/atim/xpadneo/

paroj, 2023-05-18:
https://github.com/paroj/xpad#xbox-one-controllers

closed 2023-05-18 by paroj

fbodymechanic, 2023-05-18:
I don't understand what any of that means, I was just told by Fedora helpers to make sure I let you know.  The generic xpad driver it seems like you're referencing did not work as of this morning May 18, 2023 on Fedora 38. . .

fbodymechanic, 2023-05-18:
By leaving that, does that mean you're just not planning to write the elite series controllers to work with bluetooth?

paroj, 2023-05-19:
> This driver is only used if you connect the controller via USB.