paroj/xpad #255: GameSir T4 Kaleid Support?

SaberJ2X, 2023-08-07:
Bus 003 Device 002: ID 3537:1004  Xbox360 Controller for Windows
This is the lsusb of the device

anything else I would need to provide to you guys?

Tetrajak, 2023-09-10:
I was able to get the T4 Kaleid working with the following changes in the diff below, though I don't use github enough to know how to create a PR;

```
--- xpad.c	2023-09-10 16:06:09.583078816 +1200
+++ xpad_t4k.c	2023-09-10 15:28:23.000000000 +1200
@@ -387,6 +387,7 @@
 	{ 0x31e3, 0x1310, "Wooting 60HE (ARM)", 0, XTYPE_XBOX360 },
 	{ 0x3285, 0x0607, "Nacon GC-100", 0, XTYPE_XBOX360 },
 	{ 0x3285, 0x0614, "Nacon Pro Compact", 0, XTYPE_XBOXONE },
+	{ 0x3537, 0x1004, "GameSir T4 Kaleid", 0, XTYPE_XBOX360 },
 	{ 0x3767, 0x0101, "Fanatec Speedster 3 Forceshock Wheel", 0, XTYPE_XBOX },
 	{ 0xffff, 0xffff, "Chinese-made Xbox Controller", 0, XTYPE_XBOX },
 	{ 0x0000, 0x0000, "Generic X-Box pad", 0, XTYPE_UNKNOWN }
@@ -529,6 +530,8 @@
 	XPAD_XBOX360_VENDOR(0x31e3),		/* Wooting Keyboards */
 	XPAD_XBOX360_VENDOR(0x3285),		/* Nacon GC-100 */
 	XPAD_XBOXONE_VENDOR(0x3285),		/* Nacon XBOX Series S/X Controllers */
+	XPAD_XBOX360_VENDOR(0x3537),		/* GameSir Controllers */
+	XPAD_XBOXONE_VENDOR(0x3537),		/* GameSir Controllers */
 	{ }
 };
```
The lines I added were pulled from https://github.com/torvalds/linux/blob/master/drivers/input/joystick/xpad.c

paroj, 2024-10-08:
fixed upstream

closed 2024-10-08 by paroj