paroj/xpad #260: Module not auto update after new kernel install?

Tennyleaz, 2023-09-24:
I need zeroplus controller support, so I'm using zeroplus branch: https://github.com/paroj/xpad/tree/zeroplus
Every time after a kernel update, I need to run `sudo dkms autoinstall` again to let module to build.

My system is OpenSuse TW
```
> uname -a
Linux c7h 6.5.4-1-default #1 SMP PREEMPT_DYNAMIC Wed Sep 20 05:07:04 UTC 2023 (fdd7e9e) x86_64 x86_64 x86_64 GNU/Linux

> sudo dkms autoinstall
Sign command: /lib/modules/6.5.4-1-default/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.5.4-1-default KVERSION=6.5.4-1-default...
Signing module /var/lib/dkms/xpad/0.4/build/xpad.ko
Cleaning build area...

xpad.ko.zst:
Running module version sanity check.
 - Original module
   - Found /lib/modules/6.5.4-1-default/kernel/drivers/input/joystick/xpad.ko.zst
   - Storing in /var/lib/dkms/xpad/original_module/6.5.4-1-default/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/6.5.4-1-default/updates/
depmod...
dkms autoinstall on 6.5.4-1-default/x86_64 succeeded for xpad

> sudo dkms status
xpad/0.4, 6.5.4-1-default, x86_64: installed (original_module exists)
```

I see there is `AUTOINSTALL="yes"` line already in `/usr/src/xpad-0.4/dkms.conf`. 
But seems the configuration is not working?

Or, is there any plan to add zeroplus support to mainline?

sakunix, 2023-10-06:
Hello @Tennyleaz , I use openSUSE TW too, could you help me how were you able to install xpad/zeroplus? It gives me problems:

[image]

`/bin/sh: línea 1: ./tools/bpf/resolve_btfids/resolve_btfids: No existe el fichero o el directorio
make[3]: *** [/usr/src/linux-6.5.4-1/scripts/Makefile.modfinal:62: /var/lib/dkms/xpad/0.4/build/xpad.ko] Error 127
make[3]: *** Se borra el archivo '/var/lib/dkms/xpad/0.4/build/xpad.ko'
make[2]: *** [/usr/src/linux-6.5.4-1/Makefile:1969: modules] Error 2
make[1]: *** [../../../linux-6.5.4-1/Makefile:234: __sub-make] Error 2
make[1]: se sale del directorio '/usr/src/linux-6.5.4-1-obj/x86_64/default'
make: *** [Makefile:5: all] Error 2`

make.log:
[make.log](https://github.com/paroj/xpad/files/12826360/make.log)

Tennyleaz, 2023-10-06:
Sorry I cannot understand the language in your log.

sakunix, 2023-10-06:
> Sorry I cannot understand the language in your log.

95% is English