paroj/xpad #274: 8Bitdo Ultimate Bluetooth Controller connected via 2.4 GHz dongle turns off almost instantly after connecting if it's not immediately actively used

Reonu, 2024-04-15:
Note that this doesn't happen if Steam is open, presumably because Steam takes over the controller, but that's not a solution, as I would like to use it with non-Steam apps too without having to add everything to Steam.

Also note that this is specifically about the 8Bitdo Ultimate Bluetooth Controller, and only when connected via the dongle.

So the issue is that the controller will turn off pretty much instantly after connecting if it's not being used. Even if I do maintain it connected by connecting it while a game is running, it can also turn off if, for example, I close the app that was actively interacting with the controller.

As an example, this happens with the Parallel Launcher N64 emulator.

If I manage to get in-game with the controller connected, everything is fine. But the moment I close the game, the controller will instantly disconnect (even though Parallel Launcher itself is still open). None of this happens if Steam is open in the background, it will work fine if that's the case, but that creates a new whole subset off issues.

None of this happens on Windows. There the controller stays connected until it's innactive for a few minutes and THEN it disconnects,, as expected. 

OS: EndeavourOS, up to date.

williamlagos, 2024-04-18:
I can confirm that I have the same issue here, with a 8Bitdo Ultimate Bluetooth controller, connected via USB 2.4GHz dongle to a Raspberry Pi 2B with RetroPie installed. The controller turns on, it recognizes its buttons for 6 seconds, and then turns off. My workaround for now is playing with the same controller at the same 2.4GHz mode but connected directly via USB to the RPi. I didn't tried the Bluetooth mode yet.

I'll look for system logs later to show here what is happening.

unicornsandcandies, 2024-04-19:
I can confirm that it's the issue. I'm on the newest fedora

tobiasjakobi, 2025-02-26:
FYI: This is a duplicate of this issue: https://github.com/paroj/xpad/issues/256

closed 2025-02-26 by paroj