paroj/xpad #283: Merging "support for xbox360 class controllers, that need initialisation" to upstream xpad

nondebug, 2024-08-09:
@Oblomov's PR #120 was merged 3 years ago and fixes a number of third-party Xbox 360 controllers, including Elecom JC-U4013SBK. It would be great to get these fixes upstreamed.

Is there a reason this can't be upstreamed? What needs to happen before it can be merged?

Oblomov, 2024-08-09:
FWIW the kernel driver seems to have some related changes that fix initialization of these controllers. I've been using the kernel driver instead of this module for a while now without issues.

paroj, 2024-10-18:
try changing this line to `if(0)`:
https://github.com/paroj/xpad/blob/c379818298e0e7477b4a070e219c81817412226c/xpad.c#L2008

and check whether your controller still works.

we currently send both init sequences to xpad360 controllers. What upstream does and what we did previously. I cannot verify which one is better, so I have to wait for reports.