paroj/xpad #284: New NACON Pro Contact Device ID

ronlaws86, 2024-08-16:
This device isn't picked up by xpad as it has a slight variation in the device ID, I'm guessing a different board revision out int he wild. 

ID 3285:06**03** NACON Pro Compact

DJawna, 2024-08-18:
@ronlaws86  the problem is the maintainer of this repo seems to have left the project! i had the same issue like you (i also have a nacon controller that is not in the list of ids) and i solved it by forking the project and adding my nacon controller id myself. you can see it here:
https://github.com/DJawna/xpad

ronlaws86, 2024-08-20:
> @ronlaws86 the problem is the maintainer of this repo seems to have left the project! i had the same issue like you (i also have a nacon controller that is not in the list of ids) and i solved it by forking the project and adding my nacon controller id myself. you can see it here: https://github.com/DJawna/xpad
> 
> the pr branch i did: [DJawna@21b5c6d](https://github.com/DJawna/xpad/commit/21b5c6d213ffeca77e89175b26af62b0c9f84041)

That's basically what I Did in my PR referenced above too. I've got it working, I wonder who's maintaining XPAD now? since it's included int he kernel I think? :thinking:

paroj, 2024-08-20:
> I wonder who's maintaining XPAD now?

I still intend to update this repo and merge your changes. I am just short on time right now and will need to first merge back any changes that were done in the kernel version first.

To answer your maintenance question: the xpad driver in the kernel is maintained by the kernel developers and you can send your changes there directly following this guide:
https://www.kernel.org/doc/html/v4.17/process/submitting-patches.html

However, it is not as easy as contributing on github and your changes will only appear in the next kernel release. Hence the purpose of this repo.

ronlaws86, 2024-08-20:
> I still intend to update this repo and merge your changes. I am just short on time right now and will need to first merge back any changes that were done in the kernel version first.
>
> To answer your [...]

Good to know! I marked my PR as a duplicate and closed it anyway, as it is the same as another PR submitting the same line as someone else 2 or so months ago, it can be safely discarded as it doesn't offer anything more than that one.

closed 2024-10-08 by paroj