paroj/xpad #286: xpad error and gamepad connects as Shanwan PC/PS3 android gamepad

dancaer69, 2024-08-17:
This is a wireless controller with a usb dongle 2.4GHz adapter. The device fails to connect as Microsoft X-Box 360 pad because of xpad error and falls back to this android gamepad mode. Here is the dmesg output with the error:
`[ 1085.887036] kernel: usb 3-2.3: new full-speed USB device number 15 using xhci_hcd
[ 1086.003005] kernel: usb 3-2.3: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.10
[ 1086.079412] kernel: usb 3-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1086.079550] kernel: usb 3-2.3: Product: Xbox360 For Windows
[ 1086.079665] kernel: usb 3-2.3: Manufacturer: ShanWan
[ 1086.079780] kernel: usb 3-2.3: SerialNumber: 8A2D1ED91803
[ 1086.079891] kernel: input: Microsoft X-Box 360 pad as /devices/pci0000:00/0000:00:08.1/0000:07:00.3/usb3/3-2/3-2.3/3-2.3:1.0/input/input52
[ 1086.720247] kernel: usb 3-2.3: USB disconnect, device number 15
[ 1086.750223] kernel: xpad 3-2.3:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
[ 1086.941041] kernel: usb 3-2.3: new full-speed USB device number 16 using xhci_hcd
[ 1087.054063] kernel: usb 3-2.3: New USB device found, idVendor=2563, idProduct=0526, bcdDevice= 1.00
[ 1087.140447] kernel: usb 3-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1087.140585] kernel: usb 3-2.3: Product: PC/PS3/Android GamePad
[ 1087.140715] kernel: usb 3-2.3: Manufacturer: ShanWan
[ 1087.140826] kernel: input: ShanWan PC/PS3/Android GamePad as /devices/pci0000:00/0000:00:08.1/0000:07:00.3/usb3/3-2/3-2.3/3-2.3:1.0/0003:2563:0526.0010/input/input53
[ 1087.140839] kernel: hid-generic 0003:2563:0526.0010: input,hidraw9: USB HID v1.11 Gamepad [ShanWan PC/PS3/Android GamePad] on usb-0000:07:00.3-2.3/input0`
Is this a bug? I'm using the kernel module for xpad(kernel 6.10) and I also tried to install this with git clone, but doesn't proceed because I have the exact same version.

dancaer69, 2024-08-27:
Since no respond yet I wanted to add that the device doesn't even appear in lsusb as xbox 360 controller with the id 0x45e:0x28e. Appears only as Shanwan android controller with the id 2563:0526.

P-rth, 2024-09-03:
I have a similar controller that exhibits the same issue. However, I've managed to get it working using the following method:

1. **Connect to a Windows VM:** I first connect the controller to a Windows VM running in VMware.
2. **Remove and Reconnect:** I then remove the controller from the PC and reconnect it.
3. **Xbox 360 Controller Detection:** After a few disconnections and reconnections, the controller shows up as an Xbox 360 controller.
4. **Transfer to Linux Host:** Finally, I hand over the controller to the host Linux machine by disconnecting it from the Windows VM. At this point, it stays recognized as an Xbox 360 controller and works fine.

This process suggests that the controller only detects the Linux machine during the initial connection and switches to "Android mode" as a result.

I've attached USB packet captures and `lsusb` output in a zip file for further analysis.
ive tryed to filter mos of the useless things from the usb packet capture but if these is a need for recapturing, I can do it again np
---
[lsusb-linux-windowsvm.txt](https://github.com/user-attachments/files/16855774/lsusb-linux-windowsvm.txt)
[lsusb-linux-native.txt](https://github.com/user-attachments/files/16855775/lsusb-linux-native.txt)
[captures.zip](https://github.com/user-attachments/files/16855813/captures.zip)

thank you for maintaining this awesome project

dancaer69, 2024-09-04:
I tried to disconnect/reconnect on VM a dozen times, but it doesn't work. I also tried to reset controller before reconnect, unload xpad module but never connects as xbox controller. 
I read about usb capture on other issues here and I have a windows os installed too, and I already downloaded wireshark and usb addon there, but I don't understand how to use it and identify which packets are for the controller.

SteavenGamerYT, 2026-10-07:
> 1. **Connect to a Windows VM:** I first connect the controller to a Windows VM running in VMware.
>
>     2. **Remove and Reconnect:** I then remove the controller from the PC and reconnect it.
>
>     3. [...]

IT WORKED THANKS