paroj/xpad #288: Support for 8BitDo Ultimate Wired Controller for Xbox and Windows (Hall Effect Joystick)

simpliston, 2024-09-04:
Currently, this controller is recognized as "Generic X-Box pad", but no inputs are registered. I've tested with both `jstest` and `evtest`, but pressing controller buttons don't appear in the tests.

The vendor id is `2dc8` and the product id is `2016`. It's this controller: https://shop.8bitdo.com/products/8bitdo-ultimate-wired-controller-for-xbox-and-windows

Please just ask if any more information is necessary

sprudel79, 2024-09-16:
Hi @simpliston , have you already tried [https://github.com/medusalix/xone](https://github.com/medusalix/xone)? I've been using 8bitdo ultimate wired controllers successfully for quite while, unfortunately xone only works with kernel 5.x so far.

sprudel79, 2024-09-16:
Or perhaps try a fork like [https://github.com/dlundqvist/xone ](https://github.com/dlundqvist/xone), there's a quite active development.

simpliston, 2024-10-02:
I tried them both, neither of them work for me. Using kernel version 6.10.10. I followed their install instructions, then checked in System Settings > Game Controller to see if any input would show up. It shows up on a PS5-controller, but not on this 8BitDo controller. Would it be possible to implement support here, or do you have any more pointers to something that might work?

sprudel79, 2024-10-04:
I use kernel 6.8 and have followed the instructions from the Github page:

```
git clone https://github.com/dlundqvist/xone
cd xone
sudo ./install.sh
```

Which worked for me...however, if you have an existing module installed please uninstall it first via

`sudo ./uninstall.sh`

I have the same 8bitdo controller btw. (the previous version however without hall effect update) but that should not make a difference.  Did you get any errors during the installation?