paroj/xpad #291: Failure to build with older kernels

cmitu, 2024-10-09:
Hello,

 since commit 0d861b0ddef898a68c0ed02e86d13c274d04b5b6, I cannot build the module on kernel 5.10, it fails with:

```
make -f ./scripts/Makefile.build obj=/opt/retropie/supplementary/xpad \
single-build= \
need-builtin=1 need-modorder=1
  gcc -Wp,-MMD,/opt/retropie/supplementary/xpad/.xpad.o.d  -nostdinc -isystem /usr/lib/gcc/arm-linux-gnueabihf/8/include -I./arch/arm/include -I./arch/arm/include/generated  -I./include -I./arch/arm/include/uapi -I./arch/arm/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -mlittle-endian -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -fno-dwarf2-cfi-asm -fno-omit-frame-pointer -mapcs -mno-sched-prolog -fno-ipa-sra -mabi=aapcs-linux -mfpu=vfp -marm -Wa,-mno-warn-deprecated -D__LINUX_ARM_ARCH__=7 -march=armv7-a -msoft-float -Uarm -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector-strong -Wimplicit-fallthrough -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -pg -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned  -DMODULE  -DKBUILD_BASENAME='"xpad"' -DKBUILD_MODNAME='"xpad"' -c -o /opt/retropie/supplementary/xpad/xpad.o /opt/retropie/supplementary/xpad/xpad.c
/opt/retropie/supplementary/xpad/xpad.c: In function ‘xpad_process_packet’:
/opt/retropie/supplementary/xpad/xpad.c:784:25: error: ‘ABS_PROFILE’ undeclared (first use in this function); did you mean ‘FS_NRFILE’?
   input_report_abs(dev, ABS_PROFILE, data[34]);
                         ^~~~~~~~~~~
                         FS_NRFILE
/opt/retropie/supplementary/xpad/xpad.c:784:25: note: each undeclared identifier is reported only once for each function it appears in
/opt/retropie/supplementary/xpad/xpad.c: In function ‘xpad_set_up_abs’:
/opt/retropie/supplementary/xpad/xpad.c:1809:7: error: ‘ABS_PROFILE’ undeclared (first use in this function); did you mean ‘FS_NRFILE’?
  case ABS_PROFILE: /* 4 value profile button (such as on XAC) */
       ^~~~~~~~~~~
       FS_NRFILE
/opt/retropie/supplementary/xpad/xpad.c: In function ‘xpad_init_input’:
/opt/retropie/supplementary/xpad/xpad.c:1912:30: error: ‘ABS_PROFILE’ undeclared (first use in this function); did you mean ‘FS_NRFILE’?
   xpad_set_up_abs(input_dev, ABS_PROFILE);
                              ^~~~~~~~~~~
                              FS_NRFILE
make[2]: *** [scripts/Makefile.build:280: /opt/retropie/supplementary/xpad/xpad.o] Error 1
make[1]: *** [Makefile:1825: /opt/retropie/supplementary/xpad] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.103-v7+'
```
 
 I guess there's a minimum kernel version now - do you know which one is it ?

paroj, 2024-10-09:
it was added in linux 6.1. I added a workaround with c379818298e0e7477b4a070e219c81817412226c.

That code is only used for the xbox adaptive controller.

closed 2024-10-09 by paroj

cmitu, 2024-10-09:
@paroj thank you for the fix.

cmitu, 2025-01-08:
@paroj are you still commited to allow older kernels to build the `xpad` driver ? The latest updates from upstream have added the `guard` facility which is present only after kernel 6.5. Commit d555cdce6f74e9 introduces it in `xpad.c` and breaks the build with kernels older than 6.5.

[1] https://github.com/torvalds/linux/commit/54da6a0924311c7cf5015533991e44fb8eb12773

paroj, 2025-01-08:
yes, I am currently syncing with upstream. I will explicitly revert the change here once it is done.