paroj/xpad #292: Kernel 6.11 input input33: unable to receive magic message: -32

istenburg, 2024-10-18:
Hello,
With the kernel 6.11 under arch, I recently get the following message.
```
input input33: unable to receive magic message: -32
```
What is not working properly?

paroj, 2024-10-18:
you can ignore this warning.

alternatively try changing this line to `if(0)`:
https://github.com/paroj/xpad/blob/c379818298e0e7477b4a070e219c81817412226c/xpad.c#L2008

and check whether your controller still works.

background: we currently send both init sequences to xpad360 controllers. What upstream does and what we did previously.
I cannot verify which one is better, so I have to wait for reports.

istenburg, 2024-10-19:
Thank you. The line already exists