paroj/xpad #303: Add support for GameSir Cyclone 2

hawkeye116477, 2024-12-15:
Vendor id: 3537, Product id: 1053

I tried to do that by myself, but looks like still something isn't right in x-input mode (can't test circularity at hardware tester website), similar as #243.

```
usb 1-9: new full-speed USB device number 6 using xhci_hcd
usb 1-9: New USB device found, idVendor=3537, idProduct=1053, bcdDevice= 3.28
usb 1-9: New USB device strings: Mfr=0, Product=2, SerialNumber=3
usb 1-9: Product: Xbox 360 Controller for Windows
input: GameSir Cyclone 2 as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-9/1-9:1.0/input/input35
hid-generic 0003:3537:1053.0006: ignoring exceeding usage max
input: Xbox 360 Controller for Windows Keyboard as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-9/1-9:1.1/0003:3537:1053.0006/input/input36
input: Xbox 360 Controller for Windows Mouse as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-9/1-9:1.1/0003:3537:1053.0006/input/input37
hid-generic 0003:3537:1053.0006: input,hiddev97,hidraw2: USB HID v1.10 Keyboard [Xbox 360 Controller for Windows] on usb-0000:01:00.0-9/input1

mtp-probe[4311]: checking bus 1, device 6: "/sys/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-9"
mtp-probe[4311]: bus: 1, device: 6 was not an MTP device
Watching system buttons on /dev/input/event22 (Xbox 360 Controller for Windows Keyboard)
mtp-probe[4342]: checking bus 1, device 6: "/sys/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-9"
bus: 1, device: 6 was not an MTP device
```

YogSottot, 2025-04-19:
i have gamesir cyclone 2
When connected via dongle it have different pid
In xput-mode:
```
vendor 0x3537 product 0x100b version 0x118
```

hawkeye116477, 2025-04-30:
> It is supported by xpadneo

No, I think that one only supports Xbox One controllers, so GameSir T4 Cyclone, not Cyclone 2.

YogSottot, 2025-04-30:
> No, I think that one only supports Xbox One controllers, so GameSir T4 Cyclone, not Cyclone 2.

I have and use a cyclone 2 right now and use it with xpadneo

hawkeye116477, 2025-04-30:
And how you use it with dongle? Xpadneo is only for Bluetooth. I think that you're rather simply using with xpad.
Anyway, looks like that issue with circularity test doesn't happen with Firefox. Besides that there is problem with detection of battery percentage and additional buttons not detected, but I don't know if similar fix like for Vader 4 Pro can be made :smile:.