paroj/xpad #311: Support EvoFox Elite X Wireless Gamepad

Hariprasad-max, 2025-02-14:
when plugging in the gamepad it is shown as  "ShenZhen ShanWan Technology Co., Ltd. Android Gamepad", problems  related to #119 and  #114

lsusb output: Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 001 Device 003: ID 413c:2113 Dell Computer Corp. KB216 Wired Keyboard
Bus 001 Device 004: ID 0000:3825   USB OPTICAL MOUSE
Bus 001 Device 006: ID 26ce:01a2 ASRock LED Controller
Bus 001 Device 007: ID 8087:0033 Intel Corp. AX211 Bluetooth
Bus 001 Device 073: ID 2563:0526 ShenZhen ShanWan Technology Co., Ltd. Android Gamepad
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub

weird journalctl output when trying to plug-in the gamepad -> 

[journalctl.txt](https://github.com/user-attachments/files/18798787/journalctl.txt)

lsusb -t ouput: 
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 002: Dev 002, If 0, Class=Hub, Driver=hub/4p, 480M
    |__ Port 003: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 003: Dev 003, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 004: Dev 004, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 007: Dev 073, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 007: Dev 073, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 011: Dev 006, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 014: Dev 007, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 014: Dev 007, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/10p, 20000M/x2
    |__ Port 003: Dev 002, If 0, Class=Hub, Driver=hub/4p, 5000M

lsusb -v output for 2563:0526

[lsusb-v.txt](https://github.com/user-attachments/files/18798870/lsusb-v.txt)

shans10, 2025-10-07:
I have the same controller and have the same issue. Most things work but the vibration doesn't work.

samarthhgowdaa, 2026-09-13:
Same here the vibration won't work at all