paroj/xpad #315: Broken 8Bitdo Ultimate Wireless 2.4g receiver

robtor-de, 2025-04-18:
On an old libreelec device the controller actually works, dmesg loggs the following:
```
[  118.684978] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[  118.785036] usb 1-1.3: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.14
[  118.785062] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  118.785071] usb 1-1.3: Product: Ultimate Wireless Controller
[  118.785079] usb 1-1.3: Manufacturer: 8BitDo
[  118.785086] usb 1-1.3: SerialNumber: xxxxx
[  118.809453] input: 8BitDo Pro 2 Wired Controller as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/input/input4
[  118.810908] usbcore: registered new interface driver xpad
[  118.828213] input input4: unable to receive magic message: -32
[  118.875733] input input4: unable to receive magic message: -32
[  118.964914] input input4: unable to receive magic message: -32
[  118.993331] usb 1-1.3: USB disconnect, device number 4
[  118.994970] xpad 1-1.3:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
[  119.714970] usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
[  119.810156] usb 1-1.3: New USB device found, idVendor=2dc8, idProduct=3109, bcdDevice= 2.00
[  119.810172] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  119.810176] usb 1-1.3: Product: Ultimate Wireless Controller
[  119.810180] usb 1-1.3: Manufacturer: 8BitDo
[  119.810184] usb 1-1.3: SerialNumber: xxxxx
[  119.819452] hid-generic 0003:2DC8:3109.0003: hiddev96,hidraw2: USB HID v1.11 Device [8BitDo Ultimate Wireless Controller] on usb-0000:01:00.0-1.3/input0
[  125.394020] usb 1-1.3: USB disconnect, device number 5
[  126.112808] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[  126.208648] usb 1-1.3: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.14
[  126.208664] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  126.208668] usb 1-1.3: Product: Ultimate Wireless Controller
[  126.208672] usb 1-1.3: Manufacturer: 8BitDo
[  126.208675] usb 1-1.3: SerialNumber: xxxxx
[  126.215525] input: 8BitDo Pro 2 Wired Controller as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/input/input5
```
It has kernel version 6.6.70

It is interesting that it was recogniced as the "Wired Controller" and it works:
https://github.com/paroj/xpad/commit/7130d5c8ed40aec8d7d78604a9f71ab297876f50

When using a current stable kernel it is recognized as a wireless controller but does not work, it disconnects after a few seconds:
```
[15857.481397] usb 3-2: new full-speed USB device number 10 using xhci_hcd
[15857.619254] usb 3-2: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.14
[15857.619262] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15857.619265] usb 3-2: Product: Ultimate Wireless Controller
[15857.619268] usb 3-2: Manufacturer: 8BitDo
[15857.619270] usb 3-2: SerialNumber: xxxx
[15857.621345] input: 8BitDo Ultimate Wireless / Pro 2 Wired Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/input/input31
[15857.629301] usb 3-2: USB disconnect, device number 10
[15857.629367] xpad 3-2:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
[15858.373406] usb 3-2: new full-speed USB device number 11 using xhci_hcd
[15858.518516] usb 3-2: New USB device found, idVendor=2dc8, idProduct=3109, bcdDevice= 2.00
[15858.518523] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15858.518525] usb 3-2: Product: Ultimate Wireless Controller
[15858.518527] usb 3-2: Manufacturer: 8BitDo
[15858.518528] usb 3-2: SerialNumber: xxxx
[15858.520955] hid-generic 0003:2DC8:3109.0007: hiddev2,hidraw5: USB HID v1.11 Device [8BitDo Ultimate Wireless Controller] on usb-0000:00:14.0-2/input0
[15861.066450] usb 3-2: USB disconnect, device number 11
[15861.817429] usb 3-2: new full-speed USB device number 12 using xhci_hcd
[15861.959452] usb 3-2: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.14
[15861.959460] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15861.959462] usb 3-2: Product: Ultimate Wireless Controller
[15861.959464] usb 3-2: Manufacturer: 8BitDo
[15861.959466] usb 3-2: SerialNumber: xxxx
[15861.961362] input: 8BitDo Ultimate Wireless / Pro 2 Wired Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/input/input32
[15861.974955] input input32: unable to receive magic message: -121
[15862.043966] input input32: unable to receive magic message: -121
[15866.181652] usb 3-2: USB disconnect, device number 12
[15866.181737] xpad 3-2:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
[15866.753422] usb 3-2: new full-speed USB device number 13 using xhci_hcd
[15866.895285] usb 3-2: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.14
[15866.895292] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15866.895295] usb 3-2: Product: Ultimate Wireless Controller
[15866.895296] usb 3-2: Manufacturer: 8BitDo
[15866.895298] usb 3-2: SerialNumber: xxxx
[15866.897129] input: 8BitDo Ultimate Wireless / Pro 2 Wired Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/input/input33
[15866.901357] usb 3-2: USB disconnect, device number 13
[15866.901445] xpad 3-2:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
[15867.657425] usb 3-2: new full-speed USB device number 14 using xhci_hcd
[15867.798572] usb 3-2: New USB device found, idVendor=2dc8, idProduct=3109, bcdDevice= 2.00
[15867.798582] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15867.798586] usb 3-2: Product: Ultimate Wireless Controller
[15867.798589] usb 3-2: Manufacturer: 8BitDo
[15867.798591] usb 3-2: SerialNumber: xxxx
[15867.801350] hid-generic 0003:2DC8:3109.0008: hiddev2,hidraw5: USB HID v1.11 Device [8BitDo Ultimate Wireless Controller] on usb-0000:00:14.0-2/input0

```

Another interesing observation from me was that the wireless reciver has the id 0x3109 and changes to 0x3106 when the controller is connected. Maybe this has something to do with that:
[](https://github.com/paroj/xpad/commit/0e6a710bf8d2b19e64c04ad0f6b307c27b6846bc#diff-3d985fe7837dec4db1df27733cece3b37633535ef91b1cbdf1c8b6eec4f692dbR386)

robtor-de, 2025-04-18:
I did some more investigations and when downgrading my arch kernel to 6.6.7 it still does not function.
Same when building xpad as dkms module for older commits, where the 0x3109 device id is not yet integrated. I also tried the first commit the 0x3106 module came into the sources and it did not work as well.

After view seconds I again get the usb disconnects. Maybe this is something USB related that behaves differently on the raspberry pi running LibreELEC, maybe sth power mgmt. related.

richardtatum, 2026-11-11:
I know this is an old thread, but I am still seeing this issue on kernel `7.2.14` on arch linux and it's a real headache.

Here are the logs from `journalctl` as the pad disconnects/reconnects:

```
Nov 11 21:29:18 sff kernel: usb 1-2.1: USB disconnect, device number 15
Nov 11 21:29:18 sff kernel: usb 1-2.1: new full-speed USB device number 16 using xhci_hcd
Nov 11 21:29:18 sff kernel: usb 1-2.1: New USB device found, idVendor=2dc8, idProduct=3109, bcdDevice= 2.00
Nov 11 21:29:19 sff kernel: usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 11 21:29:19 sff kernel: usb 1-2.1: Product: Ultimate Wireless Controller
Nov 11 21:29:19 sff kernel: usb 1-2.1: Manufacturer: 8BitDo
Nov 11 21:29:19 sff kernel: usb 1-2.1: SerialNumber: a6ee2fd817e4
Nov 11 21:29:19 sff kernel: hid-generic 0003:2DC8:3109.0015: hiddev97,hidraw4: USB HID v1.11 Device [8BitDo Ultimate Wireless Controller] on usb-0000:03:00.0-2.1/input0
Nov 11 21:29:20 sff kernel: usb 1-2.1: USB disconnect, device number 16
Nov 11 21:29:21 sff kernel: usb 1-2.1: new full-speed USB device number 17 using xhci_hcd
Nov 11 21:29:21 sff kernel: usb 1-2.1: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.14
Nov 11 21:29:21 sff kernel: usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 11 21:29:21 sff kernel: usb 1-2.1: Product: Ultimate Wireless Controller
Nov 11 21:29:21 sff kernel: usb 1-2.1: Manufacturer: 8BitDo
Nov 11 21:29:21 sff kernel: usb 1-2.1: SerialNumber: a6ee2fd817e4
Nov 11 21:29:21 sff kernel: input: 8BitDo Ultimate Wireless / Pro 2 Wired Controller as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-2/1-2.1/1-2.1:1.0/input/input39
Nov 11 21:29:21 sff kernel: input input39: unable to receive magic message: -121
```

Let me know if there is anything I can provide to help move this along?